kizniche/Mycodo #1423: Cant restore from backup

girrrrrrr2, 2025-05-10:
### Describe the problem/bug

When I try to restore from backup it fails to fully restore due to an issue with Marshmallow, Squalchemy and pprint? I dont know enough to really determine whats going on from there

### Versions:

 - Mycodo Version: Latest
 - Raspberry Pi Version: 4 4gb
 - Raspbian OS Version: Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.6.51+rpt-rpi-v8
    Architecture: arm64

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

Fresh install of RaspberryPI OS
Fresh install of mycodo - verified working
Restore from backup of previous instance - failure to restore

### Expected behavior

That the restore from backup function restores from a previous backup

### Additional context

Full restore log below

root@raspberrypi:/home/pi# sudo /bin/bash /opt/Mycodo/mycodo/scripts/upgrade_commands.sh backup-restore /home/pi/
.bash_history                             Desktop/                                  Pictures/                                 .wget-hsts
.bash_logout                              Documents/                                .profile                                  wiringpi_3.10_arm64.deb
.bashrc                                   Downloads/                                Public/                                   .Xauthority
Bookshelf/                                .local/                                   .sudo_as_admin_successful                 .xsession-errors
.cache/                                   Music/                                    Templates/                                .xsession-errors.old
.config/                                  Mycodo-backup-2025-05-03_13-38-27-8.16.0/ Videos/
root@raspberrypi:/home/pi# sudo /bin/bash /opt/Mycodo/mycodo/scripts/upgrade_commands.sh backup-restore /home/pi/Mycodo-backup-2025-05-03_13-38-27-8.16.0/

#### Restore of backup /home/pi/Mycodo-backup-2025-05-03_13-38-27-8.16.0/ initiated 2025-05-10_11-40-10 ####
#### Stopping Daemon and HTTP server ####
/opt/Mycodo/mycodo/scripts/mycodo_backup_restore.sh: line 39: apachectl: command not found

#### Running initialization

#### Creating mycodo user
useradd: user 'mycodo' already exists
adduser: The user `mycodo' is already a member of `adm'.
adduser: The user `mycodo' is already a member of `dialout'.
adduser: The user `mycodo' is already a member of `i2c'.
adduser: The user `mycodo' is already a member of `kmem'.
adduser: The user `mycodo' is already a member of `video'.
gpio:x:993:pi,mycodo
adduser: The user `mycodo' is already a member of `gpio'.

#### Compiling mycodo_wrapper

#### Creating symlinks to Mycodo executables

#### Creating files and directories

#### Setting permissions

#### Setting permissions

Backing up current Mycodo from /opt/Mycodo to /var/Mycodo-backups/Mycodo-backup-2025-05-10_11-40-10-8.16.0...Done.

Restoring Mycodo from /home/pi/Mycodo-backup-2025-05-03_13-38-27-8.16.0/ to /opt/Mycodo...Done.
Moving env directory...Done.

#### Running initialization

[... 390 lines cut ...]
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/opt/Mycodo/mycodo/widgets/widget_python_code.py", line 36, in <module>
    from mycodo.databases import set_uuid
  File "/opt/Mycodo/mycodo/databases/__init__.py", line 7, in <module>
    from mycodo.mycodo_flask.extensions import db
  File "/opt/Mycodo/mycodo/mycodo_flask/extensions.py", line 13, in <module>
    from flask_marshmallow import Marshmallow  # must be imported after SQLAlchemy
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/Mycodo/env/lib/python3.11/site-packages/flask_marshmallow/__init__.py", line 11, in <module>
    from marshmallow import fields as base_fields, exceptions, pprint
ImportError: cannot import name 'pprint' from 'marshmallow' (/opt/Mycodo/env/lib/python3.11/site-packages/marshmallow/__init__.py)

#### Setting permissions

#### Restarting the Mycodo daemon

#### Restarting nginx
Job for nginx.service failed because the control process exited with error code.
See "systemctl status nginx.service" and "journalctl -xeu nginx.service" for details.
#### Reloading mycodoflask

#### Connecting to http://localhost (creates Mycodo database if it doesn't exist)
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 10 times)...

kizniche, 2025-05-11:
You are attempting to restore a backup that is using a defective dependency. Notice the many import errors. This is not an issue with the restore system or Mycodo.

girrrrrrr2, 2025-05-11:
Okay I did that, it took it and it installed fine

Then went to restore the backup and it failed out at a different point, looks like its permissions?

And /var/log/mycodo/mycodorestore.log is also empty so im not gonna attach it.

```
sudo /bin/bash /opt/Mycodo/mycodo/scripts/upgrade_commands.sh backup-restore /home/pi/Documents/Mycodo-backup-2025-05-03_13-38-27-8.16.0/

#### Restore of backup /home/pi/Documents/Mycodo-backup-2025-05-03_13-38-27-8.16.0/ initiated 2025-05-10_21-14-32 ####
#### Stopping Daemon and HTTP server ####
/opt/Mycodo/mycodo/scripts/mycodo_backup_restore.sh: line 39: apachectl: command not found

#### Running initialization

#### Creating mycodo user
useradd: user 'mycodo' already exists
adduser: The user `mycodo' is already a member of `adm'.
adduser: The user `mycodo' is already a member of `dialout'.
adduser: The user `mycodo' is already a member of `i2c'.
adduser: The user `mycodo' is already a member of `kmem'.
adduser: The user `mycodo' is already a member of `video'.
gpio:x:993:pi,mycodo
adduser: The user `mycodo' is already a member of `gpio'.

#### Compiling mycodo_wrapper

#### Creating symlinks to Mycodo executables

#### Creating files and directories

#### Setting permissions

#### Setting permissions

Backing up current Mycodo from /opt/Mycodo to /var/Mycodo-backups/Mycodo-backup-2025-05-10_21-14-32-8.16.0...Done.

Restoring Mycodo from /home/pi/Documents/Mycodo-backup-2025-05-03_13-38-27-8.16.0/ to /opt/Mycodo...Done.
Moving env directory...Done.
/opt/Mycodo/mycodo/scripts/mycodo_backup_restore.sh: line 78: /opt/Mycodo/mycodo/scripts/upgrade_commands.sh: Permission denied
Failed: Error while running initialization.

Sat 10 May 21:14:51 MST 2025
#### ERROR ####
There was an error detected during the restore. Please review the log at /var/log/mycodo/mycodorestore.logpi@raspberrypi:~ $
```

kizniche, 2025-05-11:
The fix is done after the restore.

girrrrrrr2, 2025-05-11:
Okay, so I got it all the way to the point of updating to the newest version after restoring. The readings and stuff werent working but I believe you said on discord that was fixed in the newest update, so I went to update, It was stuck mid update on the site so i did it via terminal with 

sudo /bin/bash /opt/Mycodo/mycodo/scripts/upgrade_commands.sh upgrade-mycodo

Now Nginx wont start with this as its status

```× nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/lib/systemd/system/nginx.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2025-05-11 00:15:18 MST; 1min 51s ago
   Duration: 17min 51.248s
       Docs: man:nginx(8)
    Process: 8516 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=1/FAILURE)
        CPU: 19ms

May 11 00:15:18 raspberrypi systemd[1]: Starting nginx.service - A high performance web server and a reverse proxy server...
May 11 00:15:18 raspberrypi nginx[8516]: 2025/05/11 00:15:18 [emerg] 8516#8516: cannot load certificate key "/opt/Mycodo/mycodo/mycodo_flask/ssl_certs/server.key": BIO_new_file() failed (SSL: error:80000002:system library::No such file or directory:calling fopen(/opt/Mycodo/mycodo/mycodo_flask/ssl_certs/server.key, r) error:10000080:BIO routines::no such file)
May 11 00:15:18 raspberrypi nginx[8516]: nginx: configuration file /etc/nginx/nginx.conf test failed
May 11 00:15:18 raspberrypi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
May 11 00:15:18 raspberrypi systemd[1]: nginx.service: Failed with result 'exit-code'.
May 11 00:15:18 raspberrypi systemd[1]: Failed to start nginx.service - A high performance web server and a reverse proxy server.

```

kizniche, 2025-05-11:
Try running /opt/Mycodo/install/setup.sh
In the future, use the settings backup/restore to restore a system
You are finding why restoring a full backup is not a good method if your backup is in a broken state

girrrrrrr2, 2025-05-11:
That seems to have brought everything back up!

Thank you for your help and time!

closed 2025-05-11 by girrrrrrr2