kizniche/Mycodo #1426: "mycodoflask.service is not active"

OllieHydro, 2025-06-10:
### Describe the problem/bug

I have tried installing Mycodo on R pi 4 and 5, on 64 bit bookworm and 32 bit on both the pi 4 and 5. Every time i run the install line: "curl -L https://kizniche.github.io/Mycodo/install | bash" it runs through the install and eventually arrives as the same point: "mycodoflask.service is not active, cannot reload". It retries 10 times and eventually says 'mycodo installer finished...'

When i go to the device ip, i get a 403 error, when i look in the opt/Mycodo the folder is empty, no install log as mentioned in the terminal. 

I have tried not installing the database when prompted, this makes no difference.

### Versions:

 - Mycodo Version: *8.16.1*
 - Raspberry Pi Version: *4 model b and 5*
 - Raspbian OS Version: *tried 32 bit and 64 bit bookworm on both devices*

### Reproducibility

Please list specific setup details that are involved and the steps to reproduce the behavior:

1. *run the install line in terminal: curl -L https://kizniche.github.io/Mycodo/install | bash*
2. *wait until error occurs*

### Expected behavior

*the install to complete and i can continue with the setup*

### Screenshots

*[If applicable, add screenshots to help explain your problem - replace this with your own text]*

### Additional context

ive also tried manually cloning the git and installing that way, but i encounter the same problem.

OllieHydro, 2025-06-10:
i get "bash: cd /opt/Mycodo/mycodo: Permission denied"

kizniche, 2025-06-10:
Use this command, then log out, then back in, then try the commands again.

`sudo usermod -aG mycodo $USER`

pulgamecanica, 2025-06-10:
Yeah that actually is safer, didn't think about it haha, good catch kizniche, after all you need mycodo permissions

OllieHydro, 2025-06-10:
i just tried that, i get this: `noisy@raspberrypi:~ $ cd /opt/Mycodo/mycodo && sudo /opt/Mycodo/env/bin/python -m gunicorn \
--workers 1 \
--worker-class gthread \
--threads 2 \
--timeout 300 \
--pid /var/run/mycodoflask.pid \
--bind unix:/usr/local/mycodoflask.sock start_flask_ui:app
[2025-06-10 16:30:02 +0100] [1962] [INFO] Starting gunicorn 21.2.0

Error: Already running on PID 713 (or pid file '/var/run/mycodoflask.pid' is stale)

noisy@raspberrypi:/opt/Mycodo/mycodo $ kill -9 713
bash: kill: (713) - Operation not permitted
noisy@raspberrypi:/opt/Mycodo/mycodo $`

im sorry that i cant be more helpful, i simply installed the os from the raspberry pi imager onto an sd card, put the sd card in the pi, booted it, setup a new user and wifi, updated the pi, then immediately opened terminal and executed the install line to install Mycodo.

kizniche, 2025-06-10:
Let's take a step back. Try these two commands.

`sudo service mycodoflask restart`

`sudo service mycodoflask status`

kizniche, 2025-06-10:
I think there's simply a small bug that was introduced in the setup that reloads flask instead of restarting it. It seems there might not be any issue with the install at all. After those two commands, what is the output? It should be running fine and you can access the UI at https://your-Pi-IP-address, then once at least one page has loaded, restart the daemon with `sudo service mycodo restart`

kizniche, 2025-06-10:
That is expected behavior, and I can't change the message, since that is the output of the command being issued. I suppose I could change the status printed to something like "#### Disabling mycodo startup script (Note: Failing is normal if it hasn't ever been installed before)"

kizniche, 2025-06-10:
It looks like https://github.com/kizniche/Mycodo/commit/9e2c144b86d7b5c0efb28996719a8d2c83f97bff is where the issue (if this is the issue) was introduced, which changed the command from restart to reload. If @OllieHydro confirms it is actually working, I'll revert that change (as well as changing the status message about the Failure to disable).