kizniche/Mycodo #1427: display ssd1306_oled_128x64_i2c

Mark0Mi, 2025-06-16:
raspberry pi 
8.16.2

Hi
I recently updated to the latest version (jumping a lot of versions and also did an OS upgrade- which resulted in a failed update and a fresh install of mycodo ) the only thing I couldn't get to work is the display 
not 100% certain if this has to do with the new OS or the new mycodo version. 

Can you try out the following on your end -> just add the SSD 1306 128x64 
as soon as I press activate I get the following error 
according to the error there is a problem in the code - not the device - some variable not defined - thus you should be able to re-create this even if you don't have the display.

greatly appreciated any feedback I can get how to solve it.. 

2025-06-16 19:41:43,996 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_4lines_1f50b074 - Error initializing, trying again in 5 seconds: name 'DigitalInOut' is not defined
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/abstract_base_controller.py", line 56, in try_initialize
    self.initialize()
  File "/opt/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c_4lines.py", line 378, in initialize
    from mycodo.devices.lcd_pioled_circuitpython import PiOLEDCircuitpython
  File "/opt/Mycodo/mycodo/devices/lcd_pioled_circuitpython.py", line 5, in <module>
    import adafruit_ssd1306
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_ssd1306.py", line 17, in <module>
    from adafruit_bus_device import i2c_device, spi_device
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 27, in <module>
    class SPIDevice:
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 74, in SPIDevice
    chip_select: Optional[DigitalInOut] = None,
                          ^^^^^^^^^^^^
NameError: name 'DigitalInOut' is not defined
2025-06-16 19:41:49,007 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_4lines_1f50b074 - Error initializing, trying again in 5 seconds: name 'DigitalInOut' is not defined
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/abstract_base_controller.py", line 56, in try_initialize
    self.initialize()
  File "/opt/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c_4lines.py", line 378, in initialize
    from mycodo.devices.lcd_pioled_circuitpython import PiOLEDCircuitpython
  File "/opt/Mycodo/mycodo/devices/lcd_pioled_circuitpython.py", line 5, in <module>
    import adafruit_ssd1306
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_ssd1306.py", line 17, in <module>
    from adafruit_bus_device import i2c_device, spi_device
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 27, in <module>
    class SPIDevice:
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 74, in SPIDevice
    chip_select: Optional[DigitalInOut] = None,
                          ^^^^^^^^^^^^
NameError: name 'DigitalInOut' is not defined
2025-06-16 19:41:54,018 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_4lines_1f50b074 - Initialization errored 3 times; giving up. Maybe the following traceback can help diagnose the issue.
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/abstract_base_controller.py", line 56, in try_initialize
    self.initialize()
  File "/opt/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c_4lines.py", line 378, in initialize
    from mycodo.devices.lcd_pioled_circuitpython import PiOLEDCircuitpython
  File "/opt/Mycodo/mycodo/devices/lcd_pioled_circuitpython.py", line 5, in <module>
    import adafruit_ssd1306
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_ssd1306.py", line 17, in <module>
    from adafruit_bus_device import i2c_device, spi_device
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 27, in <module>
    class SPIDevice:
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 74, in SPIDevice
    chip_select: Optional[DigitalInOut] = None,
                          ^^^^^^^^^^^^
NameError: name 'DigitalInOut' is not defined

kizniche, 2025-06-16:
Have you conducted any searches for the error for the library that's causing the issue, PiOLEDCircuitpython?

kizniche, 2025-06-17:
I haven't had a chance to investigate the issue, but there's already a thread about this.

https://forum.radicaldiy.com/t/lcd-initialization-issue/1845

Mark0Mi, 2025-06-17:
Hi 
I found that I tried that and changed the config file display_ssd1306_oled_128x64_i2c.py / reloaded / 
added the 8 line display installed the dependencies 
reloaded - configured the display and activated it - still same error 

to me something basic is missing but I can't figure out what - 

2025-06-17 21:08:59,024 - INFO - mycodo - Mycodo daemon started in 13.748 seconds
2025-06-17 21:10:04,956 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_5b6028c5 - Error initializing, trying again in 5 seconds: name 'DigitalInOut' is not defined
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/abstract_base_controller.py", line 56, in try_initialize
    self.initialize()
  File "/opt/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 378, in initialize
    from mycodo.devices.lcd_pioled_circuitpython import PiOLEDCircuitpython
  File "/opt/Mycodo/mycodo/devices/lcd_pioled_circuitpython.py", line 5, in <module>
    import adafruit_ssd1306
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_ssd1306.py", line 17, in <module>
    from adafruit_bus_device import i2c_device, spi_device
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 27, in <module>
    class SPIDevice:
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 74, in SPIDevice
    chip_select: Optional[DigitalInOut] = None,
                          ^^^^^^^^^^^^
NameError: name 'DigitalInOut' is not defined
2025-06-17 21:10:09,974 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_5b6028c5 - Error initializing, trying again in 5 seconds: name 'DigitalInOut' is not defined
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/abstract_base_controller.py", line 56, in try_initialize
    self.initialize()
  File "/opt/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 378, in initialize
    from mycodo.devices.lcd_pioled_circuitpython import PiOLEDCircuitpython
  File "/opt/Mycodo/mycodo/devices/lcd_pioled_circuitpython.py", line 5, in <module>
    import adafruit_ssd1306
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_ssd1306.py", line 17, in <module>
    from adafruit_bus_device import i2c_device, spi_device
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 27, in <module>
    class SPIDevice:
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 74, in SPIDevice
    chip_select: Optional[DigitalInOut] = None,
                          ^^^^^^^^^^^^
NameError: name 'DigitalInOut' is not defined
2025-06-17 21:10:14,986 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_5b6028c5 - Initialization errored 3 times; giving up. Maybe the following traceback can help diagnose the issue.
Traceback (most recent call last):
  File "/opt/Mycodo/mycodo/abstract_base_controller.py", line 56, in try_initialize
    self.initialize()
  File "/opt/Mycodo/mycodo/functions/display_ssd1306_oled_128x64_i2c.py", line 378, in initialize
    from mycodo.devices.lcd_pioled_circuitpython import PiOLEDCircuitpython
  File "/opt/Mycodo/mycodo/devices/lcd_pioled_circuitpython.py", line 5, in <module>
    import adafruit_ssd1306
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_ssd1306.py", line 17, in <module>
    from adafruit_bus_device import i2c_device, spi_device
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 27, in <module>
    class SPIDevice:
  File "/opt/Mycodo/env/lib/python3.11/site-packages/adafruit_bus_device/spi_device.py", line 74, in SPIDevice
    chip_select: Optional[DigitalInOut] = None,
                          ^^^^^^^^^^^^
NameError: name 'DigitalInOut' is not defined
2025-06-17 21:10:14,989 - INFO - mycodo.controllers.controller_function_5b6028c5 - Activated in 10199.3 ms
2025-06-17 21:10:14,991 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_5b6028c5 - LCD not set up
2025-06-17 21:10:25,000 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_5b6028c5 - LCD not set up
2025-06-17 21:10:35,007 - ERROR - mycodo.function.display_ssd1306_oled_128x64_i2c_5b6028c5 - LCD not set up

kizniche, 2025-06-17:
I wasn't mentioning it as a solution, but rather that the same issue has already been opened elsewhere and it's not good to have multiple of the same issue opened in different places.