kizniche/Mycodo #1430: Feature Request: Add REST endpoint for remote Input-dependency installation

cajun1689, 2025-06-27:
**Is your feature request related to a problem? Please describe.**  
Mobile clients (e.g. the Ed3n iOS app) add Inputs to Mycodo via the REST API.  
When a required Python library is missing, the API currently returns an “unmet dependency” error and instructs users to log into the web UI to install it.  
This breaks the mobile workflow and confuses non-technical users.

**Describe the solution you’d like**  
Expose a secure endpoint that mirrors the web UI’s “Install dependencies” routine so apps can trigger the install directly.

---

## 📐  Proposed API

| Method | Path | Auth | JSON Body | Response 200 | Errors |
|--------|------|------|-----------|--------------|--------|
| `POST` | `/api/dependencies/install` | Bearer JWT; role `admin` **or** `install_dependencies` | Provide **one**:<br>• `input_type`: `"bh1750"`<br>• `dependency_name`: `"smbus2"` | `{ "job_id": "uuid", "state": "queued" }` | `400` bad payload<br>`403` forbidden<br>`409` identical job running<br>`500` internal |

*Job status/logs follow the existing `/api/jobs/<job_id>` pattern.*

### Security
- Uses existing JWT auth (`flask_jwt_extended`); only users with role `admin` or `install_dependencies` may call it.

### Implementation Notes
- Blueprint: `mycodo/mycodo_api/routes/dependencies.py`
- Helper: `mycodo/utils/dependencies.py`
  * Uses current `create_background_job` infrastructure to run `pip install` commands and stream logs.
- Mapping of `input_type` ➜ package list already lives in `mycodo/inputs/REQUIRED_LIBS`.
- OpenAPI docstring added so Swagger UI auto-documents the endpoint.

### Minimal example
```bash
curl -X POST https://<mycodo-host>/api/dependencies/install \
     -H "Authorization: Bearer $TOKEN" \
     -H "Content-Type: application/json" \
     -d '{"dependency_name": "smbus2"}'

kizniche, 2025-06-30:
I just added two new endpoints, /log and /dependency which will allow installing all dependencies for a particular device and tailing log lines. These aren't extensively tested and if you have any feedback or refinements, please share.

cajun1689, 2025-07-01:
🔗 Built URL: https://local ip/api/dependency/AS7341
📡 Network Request - POST https://local ip/api/dependency/AS7341
🌐 API DEBUG - API Request
📄 JSON Response:
  url: https://local ip/api/dependency/AS7341
  method: POST
  instance: Test
  hasApiKey: true

should this be how the end point is built?

kizniche, 2025-09-05:
It should be https://local.pi/api/dependency/install/device/{device_name}