kizniche/Mycodo #1435: Min-off duration off a PID not applying

Pierrevictor67, 2025-09-01:
Hello Kyle, I'm in mycodo 8.16.2

as in this post, https://github.com/kizniche/Mycodo/issues/859, I observe that the Min-off duration at raise is not applying to my situation.

I have Min-on duration set on 30 seconds. Max-on duration on 45 seconds. And Min-off duration on 600 seconds.

When I change the "Setpoint" to increase the target temperature of the PID, calculation of the PID can calculate more than 45 seconds. 30 seconds are applied several times. But there is no min-off duration  applied. And my temperature grows too fast.
On the example graph, I cut the relay manually after 4 period of 30 seconds.
Here is a graph : 

[image]

at 15:17:53 : relay on 30.188 seconds
at 15:18:23 relay on 29.831 seconds
at 15:18:53 relay on 30.006 seconds
at 15:19:23 relay on 29.464 seconds

[image]

I think you will find fast the mini bug involved.

With bets regards.
Pierre

kizniche, 2025-09-01:
It's impossible to have a min off greater than 30 seconds if your PID Period is 30 seconds.

Pierrevictor67, 2025-09-01:
> What output are you using? It's likely an issue with the output respecting the min off time, like the previous issue. You can see the PID controller provides the min off time to the output when it instructs it to turn on:
>
> https://github.com/kizniche/Mycodo/blob/96c5463feb624a777e9eb165ea1ce3f86b9cc458/mycodo/controllers/controller_pid.py#L520-L525

Thank's for your fast answer KyleThe output is a relay.
If I understand well the principle of the min off duration setting, after the PID sends an on instruction to the relay, it should be activated one time only between 30 and 45 seconds, and after that should stay off during 600 seconds.
In my case I observed a sequence of several consecutives 30 seconds for an amount superior of 45 seconds, which should not happen if the min-off duration was applied.
Am I wrong ?
With best regards.Pierre

kizniche, 2025-09-02:
The output is configured on the Setup/Outputs page and is selected in the Output (Raise) option in your screenshot.

https://kizniche.github.io/Mycodo/Supported-Outputs/

Pierrevictor67, 2025-09-03:
Hello Kule, Since I changed the different delays so that they are not multiples, I can not observe the phenomenon. I have somme 29 instead of 30 min-on duration. And 31 instead of 30 seconds for the Pid actualisation.Thank's for your several replies.
Later I will try to reproduce the phenomenon. But actually I'm in production.With best regards.Pierre

https://kizniche.github.io/Mycodo/Supported-Outputs/